elementary/panel-bluetooth #10: Duplicated bluetooth devices in dropdown menu

elementaryBot, 2017-04-07:
I often encounter this bug while pairing several bluetooth devices to my computer : one or more devices gets duplicated while they have been paired only once. The bug is persistent upon restarts and re-installations. Switchboard display is unaffected. I also noticed that states of duplicates sometimes doesn't match.
I attach screenshots of the issue.

Launchpad Details: [#LP1661536](https://bugs.launchpad.net/bugs/1661536) Antoine GRÉA - 2017-02-03 08:55:02 +0000

elementaryBot, 2017-04-07:
And this is the different state screenshot.

Launchpad Details: [#LPC](https://bugs.launchpad.net/wingpanel-indicator-bluetooth/+bug/1661536/comments/2) Antoine GRÉA - 2017-02-03 08:55:56 +0000

mdolezel, 2017-04-28:
What versions of eOS are affected by this?

naaando, 2017-04-28:
@mdolezel , I can confirm this bug exists on the lastest version _0.4 Loki_.

PR #17 (2017-06-17): Fixed: Duplicated bluetooth devices
Fixes: #10 Makes sure to only add and send the signal that a device has been added once, also prevents all devices from being deleted if you unpair a device
closed 2017-06-17 by danrabbit