elementary/panel-bluetooth #7: bluetooth disabled after suspend

elementaryBot, 2017-04-07:
Always after suspending my laptop bluetooth gets disabled and I have to activate it via wingpanel, so my bluetooth mouse gets connected.

No issues with bluetooth itself, besides it beeing switched off every time.

Launchpad Details: [#LP1637400](https://bugs.launchpad.net/bugs/1637400) Eric Göpel - 2016-10-28 05:58:12 +0000

elementaryBot, 2017-04-07:
Yes I have same issue. This drives me crazy. Very annoying if you use bluetooth mouse or keyboard.

Launchpad Details: [#LPC](https://bugs.launchpad.net/wingpanel-indicator-bluetooth/+bug/1637400/comments/1) Özgür BASKIN - 2016-10-28 18:43:51 +0000

elementaryBot, 2017-04-07:
I think this is related: #1631023

Launchpad Details: [#LPC](https://bugs.launchpad.net/wingpanel-indicator-bluetooth/+bug/1637400/comments/2) Özgür BASKIN - 2016-10-29 11:10:26 +0000

elementaryBot, 2017-04-07:
Sorry this is the correct related bug link: https://bugs.launchpad.net/wingpanel-indicator-bluetooth/+bug/1631023

Launchpad Details: [#LPC](https://bugs.launchpad.net/wingpanel-indicator-bluetooth/+bug/1637400/comments/3) Özgür BASKIN - 2016-10-29 11:12:35 +0000

elementaryBot, 2017-04-07:
This did not work for me:

https://ubuntuforums.org/showthread.php?t=1387211

Launchpad Details: [#LPC](https://bugs.launchpad.net/wingpanel-indicator-bluetooth/+bug/1637400/comments/4) Keelan Johns - 2016-11-02 03:21:17 +0000

elementaryBot, 2017-04-07:
Ozgur's comment on the duplicate bug: "Ever time I resume my laptop from suspend my bluetooth disables from wingpanel indicator. I have to enable again from wingpanel indicator for connect my mouse."

Launchpad Details: [#LPC](https://bugs.launchpad.net/wingpanel-indicator-bluetooth/+bug/1637400/comments/5) Zisu Andrei - 2016-11-03 00:54:17 +0000

elementaryBot, 2017-04-07:
Status changed to 'Confirmed' because the bug affects multiple users.

Launchpad Details: [#LPC](https://bugs.launchpad.net/wingpanel-indicator-bluetooth/+bug/1637400/comments/6) Launchpad Janitor - 2017-01-24 06:31:24 +0000

elementaryBot, 2017-04-07:
For me it's the other way around. I don't use Bluetooth and have disabled it but after suspend it's on again.

I think there was an issue about that but it maybe was closed. I don't find anything here on launchpad!

Launchpad Details: [#LPC](https://bugs.launchpad.net/wingpanel-indicator-bluetooth/+bug/1637400/comments/7) davidak - 2017-02-27 23:06:49 +0000

ruslankonev, 2017-06-17:
I have the same issue, after suspend I have a interval logging:

```
[ xxxx-xxxxxx ]Bluetooth: hci0 command 0x1003 tx timeout
[ xxxx-xxxxxx ]Bluetooth: hci0 command 0x1003 tx timeout
[ xxxx-xxxxxx ]Bluetooth: hci0 command 0x1003 tx timeout
...
```

and login page was freeze.

After kill bluetooth process, I login and  wingpanel was freeze also

queeup, 2018-01-05:
Is there any attempt to fix this issue? Since 2 years I am having this issue.

GammaGames, 2018-12-28:
Still a problem in Juno, though I did find a workaround for now (modified from [here](https://askubuntu.com/a/1085976/831221)):

1. Create a file named `bt` in `/lib/systemd/system-sleep`
2. Put the following inside the file:
```sh
#!/bin/sh

sudo service bluetooth restart
```
3. Change the script's permissions with `sudo chmod 777 /lib/systemd/system-sleep/bt`

After a few suspends to test my airpods it seems to continue working working after suspend