elementary/panel-bluetooth #9: Bluetooth indicator sync issue persists after update

elementaryBot, 2017-04-07:
Re: The 'New Year' update publicised here https://medium.com/elementaryos/loki-updates-for-the-new-year-ff6b6ba65531

This update aimed to fix the issue of the Bluetooth indicator falling out of sync with the state of the adaptor.

Having applied the update I am still seeing this issue.

Launchpad Details: [#LP1660898](https://bugs.launchpad.net/bugs/1660898) Nicholas - 2017-02-01 06:41:59 +0000

AlejandroCordova, 2017-05-06:
I have been using elementayOS for years and have never been able to use the bluetooth

danrabbit, 2018-04-09:
I'm going to close this since there are a lot of fixes in the Bluez stack in Juno that can't be backported to Loki and this issue doesn't seem to be reproducible in Juno

closed 2018-04-09 by danrabbit