elementary/panel-bluetooth #13: The bluetoot is all bugged.

GabrielMGitHub, 2017-05-03:
The bluetoot is all bugged.
When turning off the bluetooth while logged in, trying to reconnect will no longer power on.
Being logged in by turning bluetooth off and rebooting the system, it will activate again. When turning off bluetooth on the login screen and logged in, when restarting the service bluettoth does not start, the icon disappears from the bar and in system settings.

Bus 002 Device 004: ID 0cf3:3004 Atheros Communications, Inc. AR3012 Bluetooth 4.0
Bus 002 Device 003: ID 0c45:9924 Microdia 
Bus 002 Device 002: ID 8087:0024 Intel Corp. Integrated Rate Matching Hub
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 001 Device 003: ID 2232:1029 Silicon Motion WebCam SC-13HDL11939N
Bus 001 Device 002: ID 8087:0024 Intel Corp. Integrated Rate Matching Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
[    2.264635] usb 2-1.4: Product: Bluetooth USB Host Controller
[    9.573938] Bluetooth: Core ver 2.21
[    9.573960] Bluetooth: HCI device and connection manager initialized
[    9.573965] Bluetooth: HCI socket layer initialized
[    9.573968] Bluetooth: L2CAP socket layer initialized
[    9.573976] Bluetooth: SCO socket layer initialized
[   18.247205] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[   18.247210] Bluetooth: BNEP filters: protocol multicast
[   18.247215] Bluetooth: BNEP socket layer initialized
[   34.141057] Bluetooth: RFCOMM TTY layer initialized
[   34.141068] Bluetooth: RFCOMM socket layer initialized
[   34.141077] Bluetooth: RFCOMM ver 1.11
[  165.283518] Bluetooth: hci0 command 0x2007 tx timeout
[  167.288480] Bluetooth: hci0 command 0x200f tx timeout
[  169.293457] Bluetooth: hci0 command 0x2010 tx timeout
[  171.298379] Bluetooth: hci0 command 0x0c6d tx timeout

Before updating the system, the bluetooth configuration showed on and off, and visible and envisivel, now shows like this.
[image]
[image]

GabrielMGitHub, 2017-05-03:
There were 5 clean installations followed and error persisted.

GabrielMGitHub, 2017-05-03:
Problem should have been resolved here, but it was not.
https://medium.com/elementaryos/loki-updates-for-the-new-year-ff6b6ba65531

Philip-Scott, 2017-05-11:
@GabrielMGitHub Can you run `rfkill list` on the terminal, and post the output? :)

GabrielMGitHub, 2017-05-11:
@Philip-Scott I exclude this system full of failures and without support of my computer.

danrabbit, 2017-05-18:
Closing as "Incomplete" until OP provides more info

closed 2017-05-18 by danrabbit