elementary/panel-bluetooth #23: Bluetooth Shutdown Failed

GabrielMGitHub, 2017-05-27:
When turning off Bluetooth and then ending the user session, on the user login screen bluetooth still shows as active, when deactivating Bluetooth on the user login screen and restarting the system, Bluetooth returns to show as active on the screen Login. When you log on to the system, Bluetooth starts to crash.

I believe that because of the logic problem in the programming of the Bluetooth shutdown, being logged in or in the login screen, it is conflicting and consequently causing error in Bluetooth.

Portuguese:
Ao desligar o Bluetooth e em seguida encerrar a sessão do usuário, na tela de login do usuário o bluetooth ainda mostra como ativo, ao desativar o Bluetooth na tela de login de usuário e reiniciar o sistema, o Bluetooth volta a mostrar como ativo na tela login. Ao logar usuário no sistema o Bluetooth começa a apresentar falhas. 

Acredito que por problema de logica na programação do desligamento do Bluetooth, estando logando ou na tela de login, esta entrando em conflito e em consequência provocando error no Bluetooth.

GabrielMGitHub, 2017-06-08:
01:00.0 Network controller [0280]: Qualcomm Atheros AR9485 Wireless Network Adapter [168c:0032] (rev 01)
	Subsystem: Samsung Electronics Co Ltd AR9485 Wireless Network Adapter [144d:4105]
	Kernel driver in use: ath9k
	Kernel modules: ath9k, wl
Bus 002 Device 004: ID 0cf3:3004 Atheros Communications, Inc. AR3012 Bluetooth 4.0
Bus 002 Device 003: ID 0c45:9924 Microdia 
Bus 002 Device 002: ID 8087:0024 Intel Corp. Integrated Rate Matching Hub
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 001 Device 003: ID 2232:1029 Silicon Motion WebCam SC-13HDL11939N
Bus 001 Device 002: ID 8087:0024 Intel Corp. Integrated Rate Matching Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

danrabbit, 2018-04-09:
This is because Bluez doesn't keep a global memory of bluetooth status. The bluetooth indicator tries to keep a memory per user and the greeter is its own user

kp2401075, 2019-11-03:
This is still causing quite a bit of grief. Is there a possibility of a solution to this?