elementary/panel-bluetooth #29: doesn't compile successfully against (probably?) granite git master

decathorpe, 2017-08-19:
I've not been able to pinpoint the issue directly, digging through the commit logs hasn't turned up anything useful. But, the latest release fails to build against git master versions of granite (and wingpanel, for that matter), with this error message:

```
Compilation failed: 1 error(s), 1 warning(s)
src/CMakeFiles/bluetooth.dir/build.make:103: recipe for target 'src/bluetooth_valac.stamp' failed
warning: --save-temps has no effect when -C or --ccode is set
/builddir/build/BUILD/wingpanel-indicator-bluetooth-2.0.3+git170818.222502.ca2c16ea/src/Widgets/Device.vala:45.9-45.26: error: The name `get_content_widget' does not exist in the context of `Bluetooth.Widgets.Device.new'
        get_content_widget ().add (grid);
        ^^^^^^^^^^^^^^^^^^
make[2]: *** [src/bluetooth_valac.stamp] Error 1
make[2]: Leaving directory '/builddir/build/BUILD/wingpanel-indicator-bluetooth-2.0.3+git170818.222502.ca2c16ea'
CMakeFiles/Makefile2:226: recipe for target 'src/CMakeFiles/bluetooth.dir/all' failed
make[1]: *** [src/CMakeFiles/bluetooth.dir/all] Error 2
Makefile:132: recipe for target 'all' failed
make: *** [all] Error 2
```

Is it possible the bluetooth indicator is still using things that were deprecated in granite? If it helps, the nightly build on July 9 succeeded, but the build today (August 19) failed, and there were no code changes in this project at all (except the release today).

ricotz, 2017-09-01:
See elementary/wingpanel#48

decathorpe, 2017-09-03:
This issue has been resolved by the fix committed to wingpanel. Thanks!

closed 2017-09-03 by decathorpe