elementary/panel-bluetooth #64: Bluetooth icon disapeared

giovannicaligaris, 2018-10-01:
Juno

Wingpanel-indicator-bluetooth is installed. Everything works fine from switchboard as well. For some reason the icon is missing.

[image]

$ rfkill list
0: hci0: Bluetooth
	Soft blocked: no
	Hard blocked: no
1: phy0: Wireless LAN
	Soft blocked: yes
	Hard blocked: no

worldofpeace, 2018-10-07:
I noticed this also, however it was not consistently gone (after restarting wingpanel it reappeared).
Will observe if it disappears again.

Only change I see to that icon recently was https://github.com/elementary/icons/commit/03df05c821e593deab654deac372546fc5edc5c6

danrabbit, 2018-10-08:
If you reproduce the issue, can you use Gtk Inspector to see what icon name is being requested by the panel? If it's something that's not in the icon set, then we know the issue is there, but if it's `null` or empty then we know the issue is here in the indicator

rasmus91, 2018-10-12:
I'm having this same problem, occurred after coming out of suspend.

worldofpeace, 2018-10-13:
Haven't had time to dig with the gtk inspector but I've been able to reproduce this every time with `systemctl restart bluetooth`.

peteruithoven, 2018-10-19:
Any change you guys all use the wingpanel-indicator-ayatana ?
Because this issue might be related: https://github.com/elementary/wingpanel/issues/163

worldofpeace, 2018-10-19:
> Any change you guys all use the wingpanel-indicator-ayatana ?
> Because this issue might be related: [elementary/wingpanel#163](https://github.com/elementary/wingpanel/issues/163)

No, and it is realized when I look in the gtk inspector.

ggogobera, 2018-10-20:
[image]

This is happening in my case... I thought if I turned battery percentage off, it would be fixed but no difference

knorquist, 2018-10-25:
It disappears after returning from suspend for me. Sometimes the icon is completely gone, and sometimes I can still see a little like @ggogobera.

[image]
[image]

This is a fresh install so I'm not really sure what's going on.

Yansen92, 2018-10-30:
Having the same issue. Clicking the blank space still pulls up Bluetooth menu. 

[image]

peteruithoven, 2018-10-30:
If I add and exeption to the following list changing the Reveal transition to anything except SLIDE_LEFT or SLIDE_RIGHT there is no issue.
https://github.com/elementary/wingpanel/blob/master/src/Widgets/Panel.vala#L232-L251

If I switch the transition_type from right to left when making it visible again it also re-appears normally.

KadekNusa, 2018-12-28:
bluethoot indicator missing
[image]

closed 2019-01-14 by jeremypw

queeup, 2019-06-21:
I think it fixed on master. When are we going to have this fix on our systems?

peteruithoven, 2019-06-22:
You can see that the fix was merged on jan 14:
https://github.com/elementary/wingpanel/pull/187#event-2072361291
And that the latest release was on may 22: 
https://github.com/elementary/wingpanel-indicator-bluetooth/releases

So you should definitely have the fix by now.

queeup, 2019-06-22:
Hmmm I have that fix, you are right. But the problem persist on my system. It happens after wake from suspend.

peteruithoven, 2019-06-22:
@queeup could you open a new issue?
Please include as much info as possible.
For example what happens when you restart the wingpanel (`killall winganel`)?
Is there an empty space where the icon should be? Maybe include a screenshot.