elementary/panel-bluetooth #68: the bluetooth indicator disappeared

JamersonSouza, 2019-01-19:
the bluetooth indicator disappeared from my notebook for some unknown reason, when I go to check the settings, in the section of "network and wireless" the bluetooth icon is no longer there, I try to start the terminal but I can not, before that bluetooth is very time only looking for devices and when it was found it could not connect. Is there any solution to this problem?

peteruithoven, 2019-01-19:
I would have this issue on Loki in the past and ran the following commands to get it back:
```
sudo modprobe -r btusb
sudo modprobe btusb
sudo service bluetooth restart
```
To make it easier for myself I created a alias for it (really hope that's not necessary, haven't needed it on Juno):
```
alias restartbt="sudo modprobe -r btusb && sudo modprobe btusb && sleep 10 && sudo service bluetooth restart"
```

You could check the status of bluetooth using commands like:
```
service bluetooth status
rfkill list
hcitool dev
```

Suzamax, 2019-09-30:
Please consider meging #89 and this issue.

janxkoci, 2020-03-27:
I often loose access to Bluetooth (both indicator and Switchboard plug) after flipping the **Airplane mode** switch on my keyboard. At that point my wingpanel crashes (https://github.com/elementary/wingpanel/issues/248), but starts working again after a while. If I flip the Airplane mode switch some more times, I always get bluetooth back in both places..