elementary/panel-bluetooth #73: Apple wireless keyboard can't connect.

Qmo37, 2019-02-03:
Before installed elementary OS, I could connected my apple wireless keyboard to my iMac 2010 mid.
but after I installed OS, I couldn't connect to my iMac, sorry I'm new to Linux.
BTW I've tried this https://support.system76.com/articles/bluetooth/

peteruithoven, 2019-02-03:
Thanks for reporting. Could you share how you tried to connect and where it fails?
I'd also advice to research the support for the keyboard in Ubuntu, where we inherit our drivers from.

Qmo37, 2019-02-03:
Thanks for replying.
I opened Bluetooth settings, clicking pair, it just said pairing (yellow), and then failed (red).
Most of times it said available or failed no matter how I pressed the power bottom(connect bottom) on the keyboard, it still not connected.
Sorry for my poor English BTW.

peteruithoven, 2019-02-03:
I recognize this happens when I don't put a device in pairing mode, are you sure you did that?

Qmo37, 2019-02-03:
Sorry but I don't really know about the pairing mode.
Keyboard just appeared in the Bluetooth panel so I paired it.
Can you tell me more about the pairing mode?
Sorry for my stupid question.

peteruithoven, 2019-02-03:
Please look at the manual of your keyboard, this isn't a elementary OS specific thing. Usually you hold a button for a while, a light starts blinking and it's in pairing mode.

peteruithoven, 2019-02-04:
Then I'd suggest using the information available for Ubuntu to research solutions. (Because we graciously inherit the drivers from Ubuntu and their community is bigger) I've found for example:
- https://askubuntu.com/questions/730316/apple-wireless-keyboard-wont-connect-in-15-10
- https://ubuntuforums.org/showthread.php?t=963641

If you figure it out, please let us know maybe there is something we can do to improve the situation.