elementary/panel-bluetooth #82: Ability to connect a bluetooth mouse

michaelkanis, 2019-04-18:
Currently in all Linux distributions I know, there is no way of connecting a bluetooth mouse without another mouse. macOS shows a window where you can do this as it's very first thing when you set up a new Mac that has no mouse connected (like a mac mini or iMac).

I think this app would be perfect to do something similar. Would add a lot to the UX polish.

Unfortunately I can't find a screenshot for this and currently don't have access to a stationary Mac, but from what I remember the dialog was very simple. It definitely wasn't the full bluetooth prefpane. It just said something along the lines "There is no mouse connected. Connect a wired mouse or put your Bluetooth mouse into pairing mode". It would then auto pair and connect any mouse that it found and I believe it would even automatically close the dialog.

cassidyjames, 2019-04-18:
@michaelkanis luckily the Bluetooth indicator is running in all of those instances, so it could still make sense there. I've transferred this issue.

danrabbit, 2019-04-18:
Yeah bluetooth indicator probably makes sense since it's bluetooth aware and exists pretty much always

cassidyjames, 2019-04-18:
@danrabbit I guess the one counter-argument is on a system with no input devices and no Bluetooth, we might still want to prompt to like… plug in a mouse and keyboard.