elementary/panel-bluetooth #84: Bluetooth is crashing

NathanBnm, 2019-06-08:
Since the elementary OS update of May, when I try to pair a new device via Bluetooth it's just crashing. Here are the steps:

I turn the Bluetooth on,
I turn my device on,
Bluetooth indicator and plug are crashing until I turn my device off.

It seems to crash at the moment it detects a new device, very strange. But I can't use Bluetooth anymore because of this. Any idea?

peteruithoven, 2019-06-08:
Could you clarify what you mean with crash? The indicator disappears? Wingpanel disappears? 
Maybe include a screen recording. 

Could you try running Wingpanel manually: https://elementary.io/docs/code/os-dev#wingpanel
And then sharing debug logs: https://elementary.io/docs/code/os-dev#retrieving-logs
If the whole wingpanel crashes this would also be interesting: https://elementary.io/docs/code/os-dev#inspecting-crashes

NathanBnm, 2019-06-08:
Yes sure, the Bluetooth indicator only disapears. Wingpanel is not crashing.

In fact it happens at the moment I turn any new Bluetooth device on to pair it. I suppose there is an issue with new devices detection.

At this moment the Bluetooth indicator disappears. If I wait it comes back but disapears again if I try to turn Bluetooth back on and if my device is still turned off.

The Bluetooth switchboard plug is disapearing exactly the same way.

Is that more clear? I can try to make a GIF or Video capture if you want.

I'll take a look at the links you mentioned too.

peteruithoven, 2019-06-08:
This bluetooth device, has it been paired (connected before) to elementary OS already? 

It might also be interesting to start `bluetoothctl` and perform the same actions. This might give some interesting logs.

NathanBnm, 2019-06-08:
No this device (headset) wasn't paired before with elementary OS. This is working well with already-paired devices.

I'll do that

NathanBnm, 2019-06-08:
Actually this is working when I pair a device with `bluetoothctl`:

```bash
[bluetooth]# power on
[CHG] Controller A4:DB:30:79:0E:94 Class: 0x000c010c
Changing power on succeeded
[CHG] Controller A4:DB:30:79:0E:94 Powered: yes
[bluetooth]# discoverable on
Changing discoverable on succeeded
[CHG] Controller A4:DB:30:79:0E:94 Discoverable: yes
[bluetooth]# pairable on
Changing pairable on succeeded
[bluetooth]# scan on
Discovery started
[CHG] Controller A4:DB:30:79:0E:94 Discovering: yes
[NEW] Device 6A:62:9B:AD:3B:78 6A-62-9B-AD-3B-78
[CHG] Device 6A:62:9B:AD:3B:78 Name: R-Music LAÏNA

[CHG] Device 6A:62:9B:AD:3B:78 Alias: R-Music LAÏNA

[CHG] Device 6A:62:9B:AD:3B:78 ManufacturerData Key: 0x4f4d
[CHG] Device 6A:62:9B:AD:3B:78 ManufacturerData Value:
  01 00 01 fe 58                                   ....X           
[bluetooth]# pair 6A:62:9B:AD:3B:78 
Attempting to pair with 6A:62:9B:AD:3B:78
[CHG] Device 6A:62:9B:AD:3B:78 Connected: yes
[R-Music LAÏNA
[CHG] Device 6A:62:9B:AD:3B:78 UUIDs: 00001101-0000-1000-8000-00805f9b34fb
[R-Music LAÏNA
[CHG] Device 6A:62:9B:AD:3B:78 UUIDs: 0000110b-0000-1000-8000-00805f9b34fb
[R-Music LAÏNA
[CHG] Device 6A:62:9B:AD:3B:78 UUIDs: 0000110e-0000-1000-8000-00805f9b34fb
[R-Music LAÏNA
[CHG] Device 6A:62:9B:AD:3B:78 UUIDs: 0000111e-0000-1000-8000-00805f9b34fb
[R-Music LAÏNA
[CHG] Device 6A:62:9B:AD:3B:78 ServicesResolved: yes
[R-Music LAÏNA
[CHG] Device 6A:62:9B:AD:3B:78 Paired: yes
[R-Music LAÏNA
Pairing successful
[R-Music LAÏNA
[CHG] Device 6A:62:9B:AD:3B:78 ServicesResolved: no
[R-Music LAÏNA
[CHG] Device 6A:62:9B:AD:3B:78 Connected: no
[bluetooth]# connect 6A:62:9B:AD:3B:78 
Attempting to connect to 6A:62:9B:AD:3B:78
[CHG] Device 6A:62:9B:AD:3B:78 Connected: yes
[R-Music LAÏNA
Connection successful
[R-Music LAÏNA
[CHG] Device 6A:62:9B:AD:3B:78 ServicesResolved: yes
[R-Music LAÏNA
]# 
```

NathanBnm, 2020-05-01:
Seems to be resolved

closed 2020-05-01 by NathanBnm