elementary/panel-bluetooth #89: Bluetooth indicator dissapears after suspension

ezeveliz, 2019-08-08:
As the title says, the indicator dissapears after suspension, but if I click between network and battery I can see it's dropdwon and it works as it's supposed to. At least to me, it doesn't dissapear from settings pane as other issue said.
[image]
If I kill wingpanel or if I reboot it comes back.
I don't know if it's useful, but the laptop is a HP 240 G7 with a RTL8723DE chip.

maitreya2019, 2019-09-23:
I have the same issue. Reinstalling wingpanel and bluetooth indicator cannot solve it. My laptop is a Sony Vaio Pro 11.

Suzamax, 2019-09-30:
Same tho. MSI GE62 6QD with Broadcom's BTL94352Z.

ryonakano, 2019-09-30:
Closing as a duplicate of #68 because that one is posted earlier than this one.

ryonakano, 2020-03-27:
I read the author's comment of #68 again and get to feel like this issue and that one might be another one: in that issue, the author says "when I go to check the settings, in the section of "network and wireless" the bluetooth icon is no longer there", but it's not true regarding this issue. Reopening as another issue.

janxkoci, 2020-03-27:
I just want to add that for me the indicator is _sometimes_ only **partially missing**, as I explained in duplicate issue here: https://github.com/elementary/wingpanel-indicator-bluetooth/issues/106#issuecomment-604508425.

The issue is triggered for me by **suspending the laptop** (it occurs about half of the times after I login) and I was not able to trigger it in any other way.

hanaral, 2020-03-31:
I have this exact problem too, th indicator will randomly disabbear on both my macbook air and my desktop after waking them from suspend. (screenshot from my desktop)
[image]

janxkoci, 2020-05-06:
I haven't seen the issue in a few days, could it be fixed?

janxkoci, 2020-05-12:
It happened again just now :/

janxkoci, 2020-08-20:
BTW there is another tidbit that may help in resolving this - I _never_ noticed this issue in greeter, maybe the code from there can be used in normal session?

hanaral, 2021-06-05:
@janxkoci @ryonakano I haven't gotten this since Odin (a good 8 months) can any of you reproduce this anymore on beta 1?

janxkoci, 2021-06-05:
I don't have Odin but I haven't seen it on Hera too for a while now.