elementary/panel-bluetooth #106: The Bluetooth icon sometimes seems to be cut off

4jNsY6fCVqZv, 2020-03-13:
[image]

I use a stable standard system without any changes and always with the latest updates. Unfortunately I cannot say how the issue can be reproduced.

janxkoci, 2020-03-26:
I can semi-reproduce this issue to some extent - it always occurs after I suspend my laptop (by closing a lid). It doesn't appear everytime though (hence "semi-reproduce"), but I believe somewhere there is the trigger.

I also made a bunch of screenshots in the past few days with different variants of the "cut off", including "total" case:

[image]

[image]

[image]

[image]

ryonakano, 2020-03-27:
Thank you for your reports, but I'll close this issue as a duplicate of #89.

closed 2020-03-27 by ryonakano

4jNsY6fCVqZv, 2020-03-27:
@ryonakano What makes you think that disappearance and being cut off have the same source?
And what ideas do you have for solving this problem?

janxkoci, 2020-03-27:
@4jNsY6fCVqZv From my screenshots you can see that I experience both states after the same trigger. The indicator is always there, just the icon is either fully or partially missing. The underlying issue is clearly the same.

4jNsY6fCVqZv, 2020-03-27:
In this case, does it matter to you whether you have Bluetooth enabled or disabled?
For example, I never use Bluetooth.

janxkoci, 2020-03-27:
I use it sometimes for earphones. As I said, I can always fix the issue by killing wingpanel. I usually use the [Monitor app](http://appcenter.elementary.io/com.github.stsdc.monitor/) as it makes it super easy.

4jNsY6fCVqZv, 2020-03-27:
Yeah, that's not a fix in my perspective, though. It can't be that users always have to know and end their PIDs manually.

janxkoci, 2020-03-27:
> Yeah, that's not a fix in my perspective, though.

I agree.

> It can't be that users always have to know and end their PIDs manually.

I don't need PID - in the Monitor app it's easy to kill app by name, as it lists the top level parent processes. It's really neat.