elementary/panel-bluetooth #111: Devices list frequently incorrect

peteruithoven, 2020-04-16:
## Prerequisites
- [x] I have searched open and closed issues for duplicates.

## Describe the bug
The list of devices in the bluetooth indicator is freqently incorrect. 
Comparison of list in indicator with settings for example:
[image]

### To Reproduce
I'm afraid this is unclear. It happens over time. 
Instead of trying to reproduce I think it would be smarter to compare the logic in the indicator vs the settings and see if there are mistakes in the indicator. 
A first step was taken in this PR:
https://github.com/elementary/wingpanel-indicator-bluetooth/pull/17

### Expected behavior
The list of devices in the indicator is consistent with the list of devices in the bluetooth settings. 

## Platform Information
[image]

- [ ] I'm using the latest version from git that I've manually compiled
- [x] I'm using the latest released stable version

peteruithoven, 2020-05-23:
I think some issues arrise because I manually execute commands like:
- `sudo modprobe -r btusb && sudo modprobe btusb`
- `sudo service bluetooth restart`

Both of these, even individually seem to cause duplicates.  
I execute these semi frequently when I have bluetooth related issues.

peteruithoven, 2020-05-23:
I added some logs in the [Manager](https://github.com/elementary/wingpanel-indicator-bluetooth/blob/master/src/Services/Manager.vala). Looks like the `on_interface_added` is called multiple times. It's called from the `object_added` and the `interface_added` handlers. 
We need both handlers to cover all cases, but how do we prevent duplicates? Should we keep a local list of known devices and don't call the signal when it already exists?

peteruithoven, 2020-05-23:
Keeping a list of the known devices addresses and only calling the signal when it's unknown causes issues. The device status isn't properly updated in that case. 

Removing the existing device from the devices list seems to work. But only when I compared using the `address` instead of the `modalias`. (The `modalias` for both my Soundbuds Slim's was null)

closed 2020-05-27 by danrabbit

andirsun, 2020-11-04:
I still facing duplicate icon for my mouse. 
Im on 5.1.7
[image]

peteruithoven, 2020-11-04:
@andirsun did you check this with the latest version from master or just what's now in elementary? Because I don't think this has been released yet:
https://github.com/elementary/wingpanel-indicator-bluetooth/compare/2.1.5...master

astaszek, 2021-03-18:
I'm experiencing kinda the same problem after having my mouse paired to another laptop.

To reproduce :
1. Pair a BT mouse to eOS
2. Pair the mouse to another OS
3. Pair the mouse back to eOS

Expected behaviour : 
No unpaired devices in the wingpanel window, delete all previous entries related to the re-paired BT device.

Actual behaviour : 
Few old entries in the wingpanel window, few old entries in unpaired devices (which can't be even connected to, so I have to look which one is the good one).

[image]

PR #116 (2020-05-27): Prevent device duplicates
Fixes #111