elementary/panel-bluetooth #115: Mouse icon duplicate every time paired

andirsun, 2020-05-09:
## Prerequisites
- [x] I have searched open and closed issues for duplicates.

## Describe the bug
I have a BT mouse and every time i need to repair with my laptop the icon duplicates and i cant connect it throw the panel. To connect again need to open bluetooth settings, quit the current mouse icon and re pair again

[image]

### To Reproduce
Steps to reproduce the behavior:
1. Pair the mouse
2. Pair the mouse with other device
3. try to pair again with elementary
4. Icon doesnt work and need to quit an pair again

### Expected behavior

### Screenshots or screen recordings

### Logs

## Platform Information

- [ ] I'm using the latest version from git that I've manually compiled
- [x] I'm using the latest released stable version

## Additional context

danrabbit, 2020-05-29:
Should be fixed by #116

closed 2020-05-29 by danrabbit

astaszek, 2021-03-18:
I'm experiencing kinda the same problem after having my mouse paired to another laptop.

To reproduce :
1. Pair a BT mouse to eOS
2. Pair the mouse to another OS
3. Pair the mouse back to eOS

Expected behaviour : 
No unpaired devices in the wingpanel window, delete all previous entries related to the re-paired BT device.

Actual behaviour : 
Few old entries in the wingpanel window, few old entries in unpaired devices (which can't be even connected to, so I have to look which one is the good one).

[image]

[image]