elementary/panel-bluetooth #119: Bluetooth Indicator is not visible when the computer wakes up from sleep

shubhamthedev, 2020-06-22:
[image]

gorghino, 2020-09-29:
[image]

I've just got this bug. I couldn't see the bluetooth indicator in the past, but I barely saw it last time (as in the screenshot).
I clicked on 'Applications' to open GIMP and save the screenshot; I opened it while the applications list window was freezing. After some seconds the wingpanel refreshed and it restored to correct view.

Journactl:

```
set 29 13:58:25 Omen wingpanel[2772]: g_file_new_for_path: assertion 'path != NULL' failed
set 29 13:58:25 Omen wingpanel[2772]: g_file_get_uri: assertion 'G_IS_FILE (file)' failed
set 29 13:58:25 Omen io.elementary.wingpanel.desktop[2772]: **
set 29 13:58:25 Omen io.elementary.wingpanel.desktop[2772]: GLib:ERROR:../../../../glib/gvariant-serialiser.c:1366:g_variant_serialised_n_children: code should not be reached
set 29 13:58:26 Omen gimp-2.8[20141]: Impossibile trovare il motore del tema in module_path: «pixmap»,
set 29 13:58:26 Omen gimp-2.8[20141]: Impossibile trovare il motore del tema in module_path: «pixmap»,
set 29 13:58:26 Omen gimp-2.8[20141]: Impossibile trovare il motore del tema in module_path: «pixmap»,
set 29 13:58:26 Omen gimp-2.8[20141]: Impossibile trovare il motore del tema in module_path: «pixmap»,
set 29 13:58:26 Omen gimp-2.8[20141]: g_object_set_is_valid_property: object class 'GeglConfig' has no property named 'cache-size'
set 29 13:58:26 Omen gimp-2.8[20141]: gimp_device_info_set_device: assertion '(info->device == NULL && GDK_IS_DEVICE (device)) || (GDK_IS_DEVICE (info->device) && device == NULL)' failed
set 29 13:58:27 Omen gala[2602]: clutter_actor_add_effect_with_name: assertion 'CLUTTER_IS_ACTOR (self)' failed
set 29 13:58:36 Omen gnome-session[1604]: gnome-session-binary[1604]: WARNING: Application 'io.elementary.wingpanel.desktop' killed by signal 6
set 29 13:58:36 Omen gnome-session-binary[1604]: WARNING: Application 'io.elementary.wingpanel.desktop' killed by signal 6
set 29 13:58:37 Omen nm-applet[4128]: gtk_widget_destroy: assertion 'GTK_IS_WIDGET (widget)' failed
set 29 13:58:37 Omen nm-applet[4128]: gtk_widget_destroy: assertion 'GTK_IS_WIDGET (widget)' failed
set 29 13:58:37 Omen nm-applet[4128]: Can't set a parent on widget which has a parent
set 29 13:58:37 Omen nm-applet[4128]: gtk_widget_destroy: assertion 'GTK_IS_WIDGET (widget)' failed
set 29 13:58:37 Omen nm-applet[4128]: gtk_widget_destroy: assertion 'GTK_IS_WIDGET (widget)' failed
set 29 13:58:37 Omen nm-applet[4128]: Can't set a parent on widget which has a parent
set 29 13:58:37 Omen gala[2602]: invalid cast from 'CoglTexture2D' to '(null)'
```

The bug notification alert opened too on Package wingpanel 2.3.2.+r454+pkg48~ubuntu5.1.5.1. (wingpanel crashed with SIGABRT in g_assertion_message()). I've sent the report to you.

rasmus91, 2021-03-19:
I have the same issue with the Bluetooth indicator. There's an empty space on the panel. I can click it to show devices, but the icon is utterly invisible.

internetman420, 2021-04-14:
Same problem here, sort of. Suddenly the bluetooth indicator just vanished completely. Suspend/boot/restart.

ayoungethan, 2021-06-19:
This happens to me, but it's just the indicator that disappears, not bluetooth functionality. The space for the indicator remains, and if I click where the indicator should be, the indicator menu still appears.

Using this command:
`wingpanel -c bluetooth`

restores the indicator, but it appears to do so by causing a segfault and automatic reload of wingpanel.