elementary/panel-bluetooth #121: Wireless Headset disappears after suspend

zshanabek, 2020-07-15:
## Prerequisites
- [ ] I have searched open and closed issues for duplicates.

## Describe the bug
A clear and concise description of what the bug is.
After suspend I don't see available devices in bluetooth wingpanel. 

If applicable, add screenshots or screen recordings to help explain your problem.

### To Reproduce
Steps to reproduce the behavior:
1. Go to Wingpanel
2. Click on Bluetooth indicator
4. There are no available devices

### Expected behavior

I expected to see there my phone and wireless headset
### Screenshots or screen recordings

[image]

### Logs

## Platform Information

[image]

- [ ] I'm using the latest version from git that I've manually compiled
- [ ] I'm using the latest released stable version

## Additional context

andirsun, 2020-12-03:
This is not happening for me in elementary 5.1.7 or 6.0.

zshanabek, 2020-12-03:
> This is not happening for me in elementary 5.1.7 or 6.0.

And it doesn't mean that this bug is not gonna happen in other devices

closed 2021-01-29 by zshanabek