elementary/panel-bluetooth #126: Devices missing left padding with new stylesheet

cassidyjames, 2020-08-31:
So this might be a stylesheet issue, but I've only noticed it here so far. When using this indicator in the greenfield stylesheet, it looks like we're missing the side padding for devices. It looks correct under the old stylesheet.

# Hera stylesheet

[image]

# Odin stylesheet

[image]

andirsun, 2020-12-03:
This is not happening right now. Maybe you want to close the issue .

danrabbit, 2020-12-03:
Ah yeah I fixed this. It was related to images in menu menuitems and not popover menuitems :)

closed 2020-12-03 by danrabbit