elementary/panel-bluetooth #127: Bluetooth can't find any device

adrianptr, 2020-09-18:
Bluetooth can't find any device

Notebook: HP 250 G3
OS: Elementary OS 5.1.7 Hera
CPU: Intel Celeron N2840

cassidyjames, 2020-09-18:
@adrianptr the indicator just shows currently- or previously-paired devices but doesn't do discovery afaik. Can you be more specific regarding the issue you're having, what you've tried, etc.? As is there's not enough information for us to address anything here.

adrianptr, 2020-09-18:
I'm looking for new devices, it finds nothing. On a desktop computer with the same system, but with a bluetooth stylus, it works correctly. Tomorrow I will try to install third party drivers.

cassidyjames, 2020-09-18:
@adrianptr where are you looking for new devices? From System Settings → Bluetooth, or from the Bluetooth indicator in the top panel?

adrianptr, 2020-09-18:
Yes, is right section? If isn't, I'm sorry for my mistake.

Timetry, 2020-10-10:
可能你的蓝牙设备没有上电造成的，可以尝试在终端执行以下命令：
rfkill unblock bluetooth hciconfig hci0 up

adrianptr, 2020-10-11:
I resolved the problem installing third party drivers. I'm sorry for trouble. You can close the thread.

closed 2020-10-13 by tintou