elementary/panel-bluetooth #135: Bluetooth doesn't turn on

wedsonlopes, 2020-10-28:
A while ago the bluetooth stopped, and never worked again.

andirsun, 2020-12-03:
> A while ago the bluetooth stopped, and never worked again.
> 
> <bountysource-plugin>
> 
> ---
> Want to back this issue? **[Post a bounty on it!](https://www.bountysource.com/issues/93859354-bluetooth-doesn-t-turn-on?utm_campaign=plugin&utm_content=tracker%2F59641246&utm_medium=issues&utm_source=github)** We accept bounties via [Bountysource](https://www.bountysource.com/?utm_campaign=plugin&utm_content=tracker%2F59641246&utm_medium=issues&utm_source=github).
> </bountysource-plugin>

Did you solve the issue ?

kaanf, 2021-01-26:
I had a same issue after the update. Here is the solution:
`sudo apt install blueman`
Then, run Bluetooth Manager. Check for the device being trusted, and also try re-pairing in that program.