elementary/panel-bluetooth #138: Quick Pair

andirsun, 2020-12-01:
Some OSes have the feature for fast pair nearby bluetooth devices in pairing mode.

## Problem

- this PR need to be merged before work on this https://github.com/elementary/notifications/pull/15

## Proposal

The functionality follow this behavior : 
1. Put the Bluetooth peripheral in pairing mode
2. When the peripheral is close by, Windows will show a notification to the user
3. Selecting “Connect” starts pairing the peripheral
4. When the peripheral is no longer in pairing mode or is no longer nearby, the notification will be removed

## Prior Art

Windows example : 
[image]

andirsun, 2021-09-08:
Update Sep 2021: Chrome OS is implementing this feature 
[image]

cassidyjames, 2021-11-17:
@andirsun did you close this intentionally?

cassidyjames, 2021-11-17:
A note on the implementation, I believe there are three levels of integration happening here:

1. The standard way where the OS is prompting the user when a device is actively asking to pair with their computer. I'm not sure we handle this because your computer is only discoverable when the Bluetooth settings are open.
2. The pro-active (Windows) way where apparently the OS will scan for _any_ devices in pairing mode, and prompt the user when a new one shows up. This seems like it could be implemented today, but it might feel spammy and might have power usage concerns.
3. The Android/Chrome way which uses [Google's Fast Pair service](https://developers.google.com/nearby/fast-pair/spec)—as far as I can tell, a proprietary service that requires device makers to register with Google to support it, but I'm not sure if it can be implemented openly by other OSes. This does enable that more "magical" experience much like Apple AirPods but with a much larger pool of supported manufacturers and devices.