elementary/panel-bluetooth #142: Odin: bluetooth indicator is always ON in the greeter after a reboot

vjr, 2021-01-23:
## What Happened

I've disabled bluetooth in my user logged in session which seems to stay that way across reboots. However, the indicator icon in greeter login screen is always enabled and "powered" after a reboot even if I disable it in the greeter login screen.

## Expected Behavior

Indicator should remember the last set status across reboots.

## Steps to Reproduce

1. Restart computer and observe indicator being enabled in greeter login screen.
2. Disable the bluetooth indicator (powered off) via the toggle switch in the popover and restart.
3. Observe that the indicator upon restarting to greeter login screen is once again enabled by itself.

## Platform Information

elementary OS 6 Odin early access - fully updated as of filing this issue.

vjr, 2021-01-23:
I believe I have a bite size fix which I am partially testing, will open a PR soon.

Partially testing because I don't have any bluetooth devices to test with, just a desktop PC with add-on wifi+bluetooth Intel AX200 pcie card.

interroobang, 2021-02-01:
Actually I'm facing with that issue too.

vjr, 2021-04-14:
> 
> 
> Actually I'm facing with that issue too.

What wifi chipset model do you have?

I faced this issue on my desktop pci addon card with intel ax200 chip and also on my laptop with intel ax201 chip... wondering if this is an issue with the intel driver that requires working around.

interroobang, 2021-04-15:
> > Actually I'm facing with that issue too.
> 
> What wifi chipset model do you have?
> 
> I faced this issue on my desktop pci addon card with intel ax200 chip and also on my laptop with intel ax201 chip... wondering if this is an issue with the intel driver that requires working around.

Hi. I don't know which update fixed but I didn't face with issue for a while. Everything is working well now.

By the way, the devices are listed above. Thanks.

$ >lspci -knn | grep Net -A2; lsusb
02:00.0 Network controller [0280]: Qualcomm Atheros QCA6174 802.11ac Wireless Network Adapter [xxxxx] (rev 32)
	Subsystem: Dell QCA6174 802.11ac Wireless Network Adapter [1028:0310]
	Kernel driver in use: ath10k_pci
	Kernel modules: ath10k_pci
Bus 004 Device 001: ID xxxx Linux Foundation 3.0 root hub
Bus 003 Device 004: ID xxxx Realtek Semiconductor Corp. 
Bus 003 Device 006: ID xxxx A4Tech Co., Ltd. 
Bus 003 Device 005: IDxxxx Atheros Communications, Inc.

PR #143 (2021-07-20): Fix indicator status after restart or resume
Fixes #142 
Fixes #117 
Fixes https://github.com/elementary/switchboard-plug-bluetooth/issues/159

Note that I do not have any bluetooth devices to test other scenarios like pairing, disconnecting etc.

There is still the issue where the status reverts to enabled after resuming from suspend, so leaving this PR as draft for the time being to see if I can fix it in this PR itself, else will leave it for another PR.
closed 2021-07-20 by jeremypw