elementary/panel-bluetooth #154: Bluetooth blinks when toggling capslock (EOS 6 beta)

KarkanAlzwayed, 2021-05-22:
## What Happened
The Bluetooth indicator blinks when toggling the capslock button. It happens only when capslock icon is shown in the panel. I don't know about the numlock since I don't have it.

## Expected Behavior
Bluetooth indicator should mind its own business. lol

## Steps to Reproduce
Toggle capslock on and off while watch the bluetooth indicator

https://user-images.githubusercontent.com/49320606/119238118-a1989600-bb0e-11eb-8a03-20d9ae78f37a.mp4

davidmhewitt, 2021-05-23:
I imagine this is a duplicate of https://github.com/elementary/wingpanel/issues/106 ?

KarkanAlzwayed, 2021-05-23:
@davidmhewitt I'm not sure, but I know it's driving me crazy. Lol

lenemter, 2023-08-27:
Fixed in https://github.com/elementary/wingpanel/pull/486.

closed 2023-08-27 by lenemter