elementary/panel-bluetooth #155: Indicator popover shows discovered, unpaired devices when system BT scan is on

casasfernando, 2021-05-28:
## What Happened
I'm not sure if this is the right behavior or not, this is why I'm opening this issue.
I noticed this while running an application that use the system BT adapter to scan for BT BLE beacons using Bluez+D-Bus.
When the system BT adapter scan is on (normal or low energy mode), discovered but unpaired devices are listed in the indicator popover menu.
The same behavior can be reproduced by running bluetoothctl in the shell and executing the "scan on" command to start scanning for BT devices nearby.

## Expected Behavior
Only paired devices (connected or not) are listed in the indicator popover at any point in time.

## Steps to Reproduce

1. Open Terminal.
2. Execute: bluetoothctl
3. Start scanning for new devices. In bluetoothctl prompt execute: scan on
4. Wait for a few seconds while new devices are being discovered
5. Click on the BT indicator.
6. Discovered but unpaired devices are listed with the BT icon
7. To stop scanning, go back to the Terminal, and in the bluetoothctl prompt execute: scan off

### Logs

## Platform Information
elementary OS 5.1.7 Hera (up-to-date)
wingpanel-indicator-bluetooth 2.1.6+r417+pkg28~ubuntu5.1.7.1

andirsun, 2021-06-15:
I think this bug is fixes in elementary os 6. Im not able to reproduce it.
[image]
The indicator only show the paired devices.

casasfernando, 2021-06-17:
That's strange. I installed today elementary OS 6 beta 2 and I observe the same behavior.
If I go to Settings -> Bluetooth the system will start discovering devices and if while discovering I click on the indicator I see this:

[image]

All the discovered devices (but not paired) are listed in the indicator as well.
This is a new laptop with a fresh install of eOS 6 beta 2 done this morning.

peteruithoven, 2026-10-03:
Looks like this is still relevant. 
1. Open Bluetooth settings so bluetooth is in scanning mode
2. Kill the wingpanel: `killall io.elementary.wingpanel`
2. Open the bluetooth indicator

Looks like where the indicator is initially adding devices it's not filtering on whether they are paired:
https://github.com/elementary/panel-bluetooth/blob/e0fe999433f4ccc1a9133016a84155683703dd1f/src/Widgets/PopoverWidget.vala#L110

In other places it does do this paired filtering:
- https://github.com/elementary/panel-bluetooth/blob/e0fe999433f4ccc1a9133016a84155683703dd1f/src/Services/Manager.vala#L88-L90
- https://github.com/elementary/panel-bluetooth/blob/e0fe999433f4ccc1a9133016a84155683703dd1f/src/Services/Manager.vala#L96-L100

closed 2026-10-14 by danirabbit