elementary/panel-bluetooth #159: Bluetooth Won' Stay Off

Smallwheels1, 2021-06-30:
I keep the bluetooth of my computer switched off all of the time. If I Lock the computer and reactivate it, the bluetooth will switch on automatically. If I shut down the computer and restart it, the bluetooth will remain off. 

This is not something new. It has been happening for at least a year. I just haven't reported it until now. 
I'm using an Acer Aspire One laptop with a 7th generation i3 chip. 

Thank you for making Elementary great.

jeremypw, 2021-06-30:
This looks a similar issue to #142  and #117 and #23. There are three proposed fixes amongst the pull requests for this but so far none have been approved.

shawn1thompson, 2022-06-27:
This was resolved for me until I recently connected my first device. Now, it stays on again,