elementary/panel-bluetooth #183: Update indicator icon

lenemter, 2022-12-17:
### Problem

The icon uses old color and an old slashie

### Proposal

Update an icon as well as an animation

### Prior Art (Optional)

[image]

stan-janssen, 2023-04-02:
I tried to look into this yesterday. If I understand what you mean correctly: the slash should be white instead of grey, like we use on the Wifi, Audio and Notifications indicator icons.

Sure enough, the [symbolic bluetooth status](https://github.com/elementary/icons/blob/master/status/symbolic/bluetooth-disabled-symbolic.svg) icon file stil has the grey slash.

However, in the [elementary/icons](https://github.com/elementary/icons) repository, the [audio-muted](https://github.com/elementary/icons/blob/master/status/symbolic/audio-volume-muted-symbolic.svg) file also still has the grey slash, while the one in my menu bar has the white slash. 

I’m not sure if I’m looking in the right place, so: where should I be looking if these icons if I want to fix this one?

Thanks!

lenemter, 2023-04-02:
> However, in the [elementary/icons](https://github.com/elementary/icons) repository, the [audio-muted](https://github.com/elementary/icons/blob/master/status/symbolic/audio-volume-muted-symbolic.svg) file also still has the grey slash, while the one in my menu bar has the white slash.

I think the colors on wingpanel are inverted to ensure contrast, that's why you see it as white. If you use light wallpapers the slash turns grey.
[image]

lenemter, 2023-04-02:
Seems like Bluetooth icon animation is located in /status/24

jeremypw, 2023-04-02:
[image]

The slash in `audio-volume-muted-symbolic.symbolic` is actually both white and black rather than grey so its appearance will depend on the background.

jeremypw, 2023-04-02:
[image]

Compare the equivalent bluetooth icon.  The slash is grey and white.

PR #196 (2023-04-12): Gresource and update icons
Fixes #183 

These aren't fd.o icons and they're only used here so gresource them instead of getting them from the system icon theme. Namespace `indicator` to make sure we don't get the old theme icons before they're removed
closed 2023-04-12 by lenemter