elementary/panel-bluetooth #184: Bluetooth indicator

Pillot92, 2023-01-08:
### What Happened?

Bluetooth indicator on my laptop is not available

### Steps to Reproduce

I have try select bluetooth menu at setting but it's not working

### Expected Behavior

I have try select bluetooth menu at setting but it's not working. I tried to connect other device but still not work.

### OS Version

7.x (Early Access)

### Software Version

Latest release (I have run all updates)

jeremypw, 2023-01-08:
Could you whether your laptop is in "Aeroplane" mode or the bluetooth hardware is otherwise turned off or unavailable e.g. by pressing the relevant function key on the keyboard or by a hardware switch?  This makes the wingpanel icon disappear.
Setting "aeroplane" mode in System Settings does not make the icon disappear though.

vicary, 2023-06-13:
I also have this problem when testing v7 on a Raspberry Pi 4.

Things I've tried so far:
1. `rfkill unblock bluetooth` (source: [Reddit](https://www.reddit.com/r/elementaryos/comments/sxx7yc/bluetooth_missing_from_top_bar/))
2. `sudo systemctl restart bluetooth.service`
3. Rebooting
4. Software update & rebooting
5. `sudo apt update && reboot`