elementary/panel-bluetooth #199: There is no notification for the bleutooth receiving files.

saidbakr, 2023-06-14:
When sending photo from mobile phone to the PC via Bluetooth there is no any notification for the permission of receiving files so the send file fails. However, sending from the PC to the mobile phone, i.e the opposite direction, it works fine. This is also applicable with pairing the two devices.

```
~$ screenfetch
                                     said@L850
         eeeeeeeeeeeeeeeee           OS: elementary OS 5.1 hera
      eeeeeeeeeeeeeeeeeeeeeee        Kernel: x86_64 Linux 5.0.0-37-generic
    eeeee  eeeeeeeeeeee   eeeee      Uptime: 25m
  eeee   eeeee       eee     eeee    Packages: 2189
 eeee   eeee          eee     eeee   Shell: bash 4.4.20
eee    eee            eee       eee  Resolution: 1440x900
eee   eee            eee        eee  DE: GNOME 
ee    eee           eeee       eeee  WM: Mutter(Gala)
ee    eee         eeeee      eeeeee  WM Theme: elementary
ee    eee       eeeee      eeeee ee  GTK Theme: elementary [GTK2/3]
eee   eeee   eeeeee      eeeee  eee  Icon Theme: Papirus
eee    eeeeeeeeee     eeeeee    eee  Font: Open Sans 9
 eeeeeeeeeeeeeeeeeeeeeeee    eeeee   CPU: Intel Core i5-2450M @ 4x 3.1GHz [50.0°C]
  eeeeeeee eeeeeeeeeeee      eeee    GPU: AMD/ATI Thames [Radeon HD 7500M/7600M Series]
    eeeee                 eeeee      RAM: 1207MiB / 7933MiB
      eeeeeee         eeeeeee       
         eeeeeeeeeeeeeeeee          
```

jeremypw, 2023-07-07:
I am pretty sure this is fixed in elementary 7.0 (maybe in 6.1).  You need version 7.0.0 or later of the indicator.  See  https://github.com/elementary/wingpanel-indicator-bluetooth/pull/130

jeremypw, 2023-07-07:
Closing as fixed in latest version.  Please re-open if still occurring in elementary Horus.

closed 2023-07-07 by jeremypw