elementary/panel-bluetooth #203: No icon on wingpanel after updated to 7.0.0

bobby285271, 2023-06-28:
### What Happened?

As shown in the screenshot, wingpanel (the top panel) is missing a bluetooth icon:

[image]

### Steps to Reproduce

1. Boot to NixOS https://github.com/NixOS/nixpkgs/pull/240261/commits/0afd3aacbec7bd21ecdd3a060b0674786e2a6331
2. See the wingpanel?

### Expected Behavior

Icon appears.

### OS Version

Other Linux

### Software Version

Latest release (I have run all updates)

bobby285271, 2023-06-28:
I am looking at https://github.com/elementary/wingpanel-indicator-bluetooth/pull/196 since this is probably the only icon-related change in this bump.

In https://github.com/NixOS/nixpkgs/pull/240261 I am also testing a wingpanel-indicator-notifications bump, which has very [similar change](https://github.com/elementary/wingpanel-indicator-notifications/commit/c4db1878d49b7c9e07dee99827802b7a42102ebf), but it looks like the wingpanel-indicator-notifications icon still works fine.

bobby285271, 2023-06-28:
Now I am looking these lines:

https://github.com/elementary/wingpanel-indicator-notifications/blob/c4db1878d49b7c9e07dee99827802b7a42102ebf/src/Indicator.vala#L50

and

https://github.com/elementary/wingpanel-indicator-bluetooth/blob/8d8ca94d48f9234e6ab81303ddabe77d0885d3fa/src/Widgets/DisplayWidget.vala#L29

Not sure if "when these are loaded" matters, going to test after lunch.

bobby285271, 2023-06-28:
I found the icon actually appears fine in greeter/lock screen, it is just missing in a desktop session :thinking:

[image]

Though I don't think this is caused by messy environment variables on my personal account (:upside_down_face:) since I also tried this on a fresh VM (by forcing the indicator to be visible).

[image]

tintou, 2023-06-28:
@danirabbit I believe that we are missing `Gtk.IconTheme.get_default ().add_resource_path ("/org/elementary/wingpanel/icons")` at some place.

Also, `<gresource prefix="org/elementary/wingpanel/icons">` is missing the leading slash

tintou, 2023-06-28:
I would personally change it for `/io/elementary/wingpanel/bluetooth/icons` to avoid prefix clash

danirabbit, 2023-06-28:
Hm I'm not able to reproduce this.

@tintou It's namespaced that way intentionally so that Gtk.Application loads the resource into the icon path automatically and don't need to manually add it

tintou, 2023-06-28:
Ah true, we still need the leading slash though

jeremypw, 2023-06-28:
Perhaps the automatic loading of the resource to the icon theme is not happening on NixOS for some reason.  This issue does not occur on elementary.

bobby285271, 2023-06-28:
Yeah from my testing a explicit add_resource_path does fixes the issue for me, i.e.

```patch
diff --git a/src/Widgets/DisplayWidget.vala b/src/Widgets/DisplayWidget.vala
index daded09..bb7ce5e 100644
--- a/src/Widgets/DisplayWidget.vala
+++ b/src/Widgets/DisplayWidget.vala
@@ -25,6 +25,9 @@ public class BluetoothIndicator.Widgets.DisplayWidget : Gtk.Spinner {
     }
 
     construct {
+        weak Gtk.IconTheme default_theme = Gtk.IconTheme.get_default ();
+        default_theme.add_resource_path ("/org/elementary/wingpanel/icons");
+
         var provider = new Gtk.CssProvider ();
         provider.load_from_resource ("io/elementary/wingpanel/bluetooth/indicator.css");
```

And from testing it sounds like doing this also helps other indicators, i.e. after adding this patch I found https://github.com/elementary/wingpanel-indicator-nightlight/pull/101#issuecomment-1611732534 fixed too.

/sos @jtojnar :upside_down_face:

tintou, 2023-06-28:
@danirabbit I believe that the autoloading of resources is done too early in the process when the indicators are not already loaded, and as they are plugins it might be the case, I propose to use some explicit loading to be on the safe side even in elementary

danirabbit, 2023-06-28:
Sounds good, I'll make a branch!

PR #204 (2023-06-29): Manually add resourced icons to path
Fixes #203