elementary/panel-bluetooth #206: Battery indicator for bluetooth devices

mar0425, 2023-07-04:
### Problem

Hey,

is it possible to display battery levels on bluetooth devices (like headphones, speakers) like other OS do?

would be so useful... thanks 

### Proposal

Display it next to the listed devices when you click on the bt indicator

CodingSpiderFox, 2023-11-02:
This would be a really really great feature. 

The most annoying thing when having a meeting is joining and having your headphones turn off in the middle of the meeting.

Depending on the device used, it might not be possible to keep using them during charge. So It would also be nice if a notification with high importance is displayed to the user when the battery percentage drops below a user-chosen threshold for any of the currently connected bluetooth speakers / headphones like it is already done when the laptop battery is almost empty

danirabbit, 2023-11-06:
Just want to make a note that devices which communicate their battery status will appear in the power indicator

CodingSpiderFox, 2023-11-06:
I created #210 for the notification because I think this is a different feature request than this one here

CodingSpiderFox, 2023-11-06:
> Just want to make a note that devices which communicate their battery status will appear in the power indicator

I tested this with my Bose QC35 II.
They report the battery level on my iPad correctly but when connecting them to my laptop with the latest elementaryOS version, I can only see the laptop battery level when I click the battery icon in the top panel

danirabbit, 2023-11-06:
@CodingSpiderFox this feature uses UPower, so if you open up D-Spy and search for `upower` on the system bus tab you can see a list of devices exposed by UPower there.

[image]

If your headphones _do_ show up in this list but not the power indicator, please open an issue report about that and we can try to figure out what's happening. But if they don't show up in that list, you'll need to report an issue upstream against UPower here: https://gitlab.freedesktop.org/upower/upower/-/issues

mar0425, 2023-11-06:
Tested this, none of my devices (tested with three bluetooth headphones and a bluetooth keyboard) show on d-spy under upower, they don´t show in the indicator either, all of them are supposed to be comunicating battery levels (I can see them on my android phone, for example) ... maybe it's an issue with upower?

danirabbit, 2023-11-07:
@mar0425 Yes, if your devices aren't showing up under D-Spy, please file an issue against UPower on GitLab: https://gitlab.freedesktop.org/upower/upower/-/issues

If UPower exposes the device over DBus that's when we can show it in the panel

lenemter, 2023-11-07:
You may want to look at https://askubuntu.com/questions/1117563/check-bluetooth-headphones-battery-status-in-linux. It helped me a while ago.

danirabbit, 2023-11-07:
Ah that's good to know it's a Bluez issue and not UPower. It looks like this was marked as non-experimental last month :tada: https://git.kernel.org/pub/scm/bluetooth/bluez.git/commit/?id=495a582675cf0a4b2fee293cc215504ef24a1c43

So we'll have to see when Bluez 5.71 is released and if that's imported in Noble and then it should be fixed :)

mar0425, 2023-11-08:
Ended up activating bluez experimental features following @lenemter 's post and my devices are showing now on power menu, I'll keep that config until bluez next release